12urenloop/cvc #75: Automatic Twitterwall

Silox, 2015-02-12:
There was an idea to let something (Boxxy?) post automatich twitter messages when someone overtakes someone, when someone reaches a certain amount of laps, when someone ran X-kilometers etc.

Silox, 2015-02-12:
This could be used to communicate to twitter https://www.npmjs.com/package/twitter

FelixVanderJeugt, 2015-02-13:
If that which you linked is a javascript API to twitter, then yes, this should be done with boxxy. It however says it's a node.js client. That should still work, but might be a bit harder.

Silox, 2015-02-13:
It is possible to POST messages, yes: https://dev.twitter.com/rest/reference/post/statuses/update

Procrat, 2016-02-22:
We should review the code.

Iepoev, 2016-03-31:
I've reviewed the output and most of the features work as expected.
Lap speed records however are broken because the data is way too inaccurate to keep track of lap speed

Procrat, 2016-04-06:
@Iepoev: Cool! So can you remove that part? Then that's one less thing we have to worry about! ^_^

closed 2018-03-09 by iliclaey