12urenloop/cvc #89: Configure all the gyrids

Silox, 2015-04-02:
Make sure everything works; try to put them in a circle

Label them from 101 to 107 (or 108?) and give them a corresponding ip adress (x.y.z.101 eg)

Silox, 2015-04-03:
* 108: upgrading went wrong here, python was wobbly, abort abort
* 109: OK
* 110: to do
* 121: OK
* 102: to do
* 113: to do
* 330: should be fine (Has a sticker: Valt uit?) but noticed no problems here

Silox, 2015-04-03:
Everything has network ... Let's try to complete the loop now.

Silox, 2015-04-03:
Yesh, it works. The loop is there, ports seem to do something. 

[image]

Procrat, 2016-02-25:
Reopening this for 2016.

NoctuaNivalis, 2016-02-26:
Note that we should be able to flash all gyrids cards with the device dump we've taken and used last year.

Procrat, 2016-03-23:
Note for next year: to connect with the serial port, you can use one of [these programs](https://wiki.archlinux.org/index.php/Working_with_the_serial_console) with baud rate 24000, 8 data bits, no parity bit and 1 stopbit.

mivdnber, 2016-03-23:
Are you sure about that? I always connect at 38400 baud. Easiest is `screen /dev/<PORT> 38400`.

Silox, 2016-03-30:
2016:

All Connecting to: 10.1.2.178:2583

101: should be ok
102: should be ok
103: should be ok
105: should be ok
112: should be ok
unlabeled 1: no idea
unlabeled 2: no idea

Silox, 2016-03-30:
Current status: gyrids should be configured correctly and are all booting. 
Next todo: check if they actually connect with CVC and send data

Silox, 2016-03-30:
Fixed the /usr/share/gyrid/network_middleware to s/caching/cache/ on line 675 or something. **NOT DONE FOR 102 YET!!!**

Iepoev, 2016-03-31:
finished 102 config

Iepoev, 2016-03-31:
on the topic of stp:

the problem was that the gyrids all have the same bridge priority id (because we are just copypasting the `config/interfaces` file). However, the id in this file is/was lower than the default id of the cisco switch. This resulted in all of the gyrids getting into an ex aequo when electing the root. The solution was to lower the bridge priority id of the switch to 0.

PR #136 resolves this by increasing the bridge priority id of the `config/interfaces` to the maximum value

becousae, 2016-12-12:
Reopen this for 2017.

Iepoev, 2017-04-10:
all gyrids have been labeled and have been assigned a correct hostname + ip address
they still need to be patched
https://github.com/ZeusWPI/12urenloop/issues/114#issuecomment-206234613

and  `hwclock -w` has also not been run

NoctuaNivalis, 2017-08-07:
2017 edition is over. Also kindly open a new issue (with a link here) next year, this issue keeps showing up in my TODO.

closed 2017-08-07 by NoctuaNivalis