12urenloop/cvc #90: Configure the network

Silox, 2015-04-02:
We have two incoming links from the abdij, these go both into Token (one main and one fallback interface)

One cable goes out of token into the switch

We need the following vlans on this switch:

1. Urgent
  * One cable to their streaming system
  * One cable to a wifi router on their stage (small one)
2. Counting system
  * One cable to Felix' laptop
  * One to both gyrids at the "ends" of the loop
3. Our own wifi and general network
  * Wifi router (kelder main one)
  * Materiaalbeheersysteem
  * ...

Silox, 2015-04-03:
Switch ports:

VLANs:

1. Counting system
2. Urgent
3. Coordinatie

 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12
 ----|----|----|----|----|----|----|----|----|----|----|----
 UP | - | CS | CS | CS | - | URG | - | 9 | CO | CO | CO 

 13 | 14 | 15 | 16 | 17 | 18 | 19 | 20 | 21 | 22 | 23 | 24
 ----|----|----|----|----|----|----|----|----|----|----|----
 UP | - | CS | CS | CS | - | URG | - | 9 | CO | CO | CO

Procrat, 2016-02-25:
Reopening for 2016.

Iepoev, 2016-03-31:
forgot about token

Iepoev, 2016-04-12:
```  Interface Administrative VLANs
  GE1 1T, 4095P  
  GE2 1UP
  GE3 1UP
  GE4 3UP
  GE5 3UP
  GE6 5UP
  GE7 5UP
  GE8 1UP
  GE9 4UP
  GE10  4UP
  GE11  4UP
  GE12  4UP
  GE13  1T, 4095P  
  GE14  1UP
  GE15  3UP
  GE16  3UP
  GE17  1UP
  GE18  5UP
  GE19  5UP
  GE20  1UP
  GE21  4UP
  GE22  4UP
  GE23  4UP
  GE24  4UP
  GE25  1UP
  GE26  1UP```

Silox, 2016-04-12:
Should work fine.

closed 2018-03-09 by iliclaey