12urenloop/cvc #114: Gyrid Code

FelixVanderJeugt, 2015-04-30:
Find/create a recent public version of the Gyrid code.

mivdnber, 2015-04-30:
Hold your horses there, cowboy. I'm not sure if certain people at UGent would like the current gyrid code floating around. I've got a meeting with all relevant people on monday; I can ask them for permission.

FelixVanderJeugt, 2015-04-30:
In case they don't like it, let's substitute "public" to "private", and put it in a private Zeus git repo? I would prefer to have some version of the gyrid code (which we've used before) available for next year.

FelixVanderJeugt, 2015-04-30:
Note: the version which is currently available on github is GnuGPL licenced.

Procrat, 2016-02-17:
@mivdnber: What was the result of this meeting?

mivdnber, 2016-02-23:
It's been a while, but I don't think there were any objections.

Procrat, 2016-02-24:
Awesome! :D

Procrat, 2016-04-06:
Reminder to also add [this patch](https://github.com/ZeusWPI/12urenloop/issues/89#issuecomment-203651729) when we put this in a repo.

Silox, 2016-04-06:
Actual patch:
```
diff --git a/network_middleware.py b/network_middleware.py
index 4f89014..d8f7b32 100644
--- a/network_middleware.py
+++ b/network_middleware.py
@@ -564,7 +564,7 @@ class InetClient(Int16StringReceiver):
             elif msg.requestCaching.clearCache:
                 self.clearCache()
 
-            self.factory.config['enable_caching'] = msg.requestCaching.enableCaching
+            self.factory.config['enable_cache'] = msg.requestCaching.enableCaching
 
             msg.success = True
             self.sendMsg(msg, await_ack=False)
```