12urenloop/cvc #119: Add historiek to site

Silox, 2015-06-28:
EDITIE	JAAR	DATUM	WINNAAR
1	1984	7 maart	Rijks Normaal School
2	1985	13 februari	RNS
3 	1985	13 november	RNS
4	1986	29 oktober	VTK
5 	1987	28 oktober	VTK
6	1988	26 oktober	Koninklijke Militaire School
7	1989	25 oktober	VTK
8			VTK
9	1992	19 februari	HILOK
10			HILOK
11			
12			
13 	1995	8 november	VTK
14	1996		HILOK
15	1997	26 november	
16	1999		
17	2000	25 oktober	VTK
18	2001	24 oktober	HILOK
19	2002	23 oktober	
20	2003	22 oktober	
21	2004	27 oktober	HILOK
22	2006	1 maart	
23	2007	28 februari	HILOK
24	2008	8 april	VTK
25	2009	4 maart	VTK
26	2010	21 april	VTK
27	2011	4 mei	HILOK
28	2012	25 april	HILOK
29	2013	24 april	VTK
30	2014	30 april	VTK
31	2015	29 april	VTK

lege dingen = geen info/verloren info

FelixVanderJeugt, 2015-06-28:
Kunnen we daar met gerust hart geen VTK of HILOK in invullen?

Silox, 2015-06-28:
Neen

Procrat, 2015-07-14:
1997: HILOK. (Source: http://www.schamper.ugent.be/files/pdf/354.pdf)

Procrat, 2015-07-15:
2008: was op 9 april, niet 8 april (Source: http://www.schamper.ugent.be/files/pdf/462.pdf)

Procrat, 2015-07-15:
1998: geen organisatoren voor 12urenloop gevonden (Source: http://student.ugent.be/akcentraal/rp19981126.html)

Procrat, 2015-07-15:
1999: 17 november. (Source: http://www.schamper.ugent.be/files/pdf/373.pdf)

Procrat, 2015-07-15:
1992: vond plaats op 21 oktober, niet 19 februari (Source: http://www.schamper.ugent.be/files/pdf/298.pdf)

Procrat, 2015-07-15:
1991: 19 februari (Source: http://www.schamper.ugent.be/files/pdf/295.pdf)

Procrat, 2015-07-15:
1990: 7 november. VTK is gewonnen (Source: http://www.schamper.ugent.be/files/pdf/280.pdf)

Procrat, 2015-07-15:
Update

Jaar | Datum | Winnaar
---- | ----- | -------
1984 | 7 maart | Rijks Normaal School
1985 | 13 februari | RNS
1985 | 13 november | RNS
1986 | 29 oktober | VTK
1987 | 28 oktober | VTK
1988 | 26 oktober | Koninklijke Militaire School
1989 | 25 oktober | VTK
1990 | 7 november | VTK
1991 | 19 februari | ???
1992 | 21 oktober | HILOK
1993 | ??? | ???
1994 | ??? | ???
1995 | 8 november | VTK
1996 | ??? | HILOK
1997 | 26 november | HILOK
1998 | (niet doorgegaan)
1999 | 17 november | ???
2000 | 25 oktober | VTK
2001 | 24 oktober | HILOK
2002 | 23 oktober | ???
2003 | 22 oktober | ???
2004 | 27 oktober | HILOK
2006 | 1 maart | ???
2007 | 28 februari | HILOK
2008 | 9 april | VTK
2009 | 4 maart | VTK
2010 | 21 april | VTK
2011 | 4 mei | HILOK
2012 | 25 april | HILOK
2013 | 24 april | VTK
2014 | 30 april | VTK
2015 | 29 april | VTK

bmesuere, 2015-07-15:
1991: HILOK (http://www.schamper.ugent.be/files/pdf/296.pdf)

bmesuere, 2015-07-15:
1994: 9 november (http://www.schamper.ugent.be/files/pdf/323.pdf)

Procrat, 2015-07-15:
Ik ga de jaartallen eens aanpassen naar academiejaren, want dit is verwarrend: die van 1991 die je net poste, @bmesuere, viel eigenlijk in 1992. :D

bmesuere, 2015-07-15:
oei, sorry!

Procrat, 2015-07-15:
04-05: HILOK (Source: https://web.archive.org/web/20061216184552/http://fkserv.ugent.be/~urenloop/results.php)

Procrat, 2015-07-15:
99-00: VTK (Source: http://www.nieuwsblad.be/cnt/nbrb26102001_002)

JulienMar, 2016-01-03:
UGentMemorie over de 12urenloop:

Op 7 maart 1984 wordt ter gelegenheid van het 75-jarig bestaan van het Hoger Instituut voor Lichamelijke Opvoeding (HILO) voor het eerst een 12-urenloop gehouden. Tijdens deze jaarlijks terugkerende manifestatie lopen talrijke sportievelingen (zowel studenten als personeel) gedurende een half etmaal in estafette rond het Sint-Pietersplein. De happening omvat nog andere studentikoze spelen en optredens. Aanvankelijk organiseert het Activiteiten Konvent de 12-urenloop maar na verloop van tijd neemt het Faculteiten Konvent de sportmanifestatie over. De Twaalfurenloop evolueert van een sportieve studentenbijeenkomst naar een echte studentikoze gebeurtenis met zijn eigen tradities: zo is er altijd veel bier, ludieke rondes waarbij de deelnemers bijvoorbeeld naakt of verkleed moeten lopen. Ook traditioneel is de top-2: het Hilok of het VTK, de verenigingen van de studenten Lichamelijke Opvoeding en van de ingenieursstudenten, schrijven beurtelings de meeste rondjes op hun palmares.