12urenloop/cvc #128: Fix internet

Procrat, 2016-02-25:
- [ ] Trying Digipolis
- [ ] Try ISPs
- [ ] Try using wireless internet from GUSB

NoctuaNivalis, 2016-02-26:
I am in favor of lasering internet across the river.

Procrat, 2016-02-27:
Could you elaborate, @NoctuaNivalis?

NoctuaNivalis, 2016-02-28:
Sure thing @Procrat, there you go:
```
  .------.
  | GUSB |            LASER             .-----------.
__|      >---------o----o--------o----->| CONTAINER |_______
         |                              |
         |~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~|
         |        WATERSPORTWATER       |
         '------------------------------'
```

Procrat, 2016-03-02:
We received some lasers to test with. ^^

becousae, 2016-03-07:
Testing the lasers on s9 was successful. Nobody was killed.

NoctuaNivalis, 2016-03-08:
And how exactly is that successful?

Procrat, 2016-04-06:
- Both DICT and the people responsible at GUSB were very wary. They also suggested asking rowing clubs or doing it with a 4G router anyway.
- The rowing club at the right side of the water was closed when I went and is apparently only open two hours in a week.
- There's a cabin next to the rowing club that is lent to personal trainers. I got the contact info of the owner of these cabins. I will contact him later today.

Procrat, 2016-04-06:
As for the ISPs:
- Belgacom (actually Proximus) gave us an offer of €58 (excl. btw) for the subscription + €59 for the installation.
- Telenet didn't reply, even after a reminder mail.

Procrat, 2016-04-06:
I phoned the owner of the personal trainer cabin. We still have to work out the details, but he seemed quite willing to share his internet connection (maybe in exchange for some advertising).
I'll update as soon as I hear more.

Procrat, 2016-04-12:
Leander Verbraeken is an awesome guy!

closed 2016-04-12 by Procrat