12urenloop/cvc #131: What with internet loss

Procrat, 2016-02-25:
Should we worry about this?

Silox, 2016-02-25:
Yes, not necessarily for us (unless Saruman is hosted off-site), but it's important for the Urgent stream.

NoctuaNivalis, 2016-02-26:
Well, if we lose connection, the live updates on the site wouldn't be live any longer. That's a shame, but no big trouble (though perhaps we should put a notice on the site saying the connection is lost?).

Procrat, 2016-02-29:
I don't think we should worry too much about long-term internet loss. In case of emergency, we can always hack something together with the 4G modem, I guess?

I was hinting about temporary internet loss. So, to conclude
- for the counting system: Boxxy doesn't get new updates, but no biggie. @NoctuaNivalis: I'll make an issue for the notice.
- for Saruman: @Silox: Is it correct that there is no solution for this except running multiple instances of Saruman? If it's hosted locally, partners can't access it and if it's hosted on King, we can't manage pickups anymore, right? Should we freeze the database or something like that or should we not bother?
- for Urgent: Too bad.
- for public wifi to live.12urenloop.be (this is new, not certain yet): Too bad. (We could also set up a local Boxxy server, but this is probably more work than it's worth.)

Any more comments on this? Also paging @Iepoev and @becousae.

NoctuaNivalis, 2016-02-29:
Settings up a local boxxy server is not a lot of work. Just installing npm and it should "work."

closed 2016-04-09 by Procrat