12urenloop/cvc #133: Material for 2016

Procrat, 2016-02-25:
*Network*
- [ ] Managed switch (of SKO, @Iepoev)
- [ ] Router (Token)
- [ ] Two Wifi APs (both of Zeus)
- [ ] Network cable
- [ ] Network cable making stuff: connectors, cable shields, pliers, tester

*PCs and stuff for us & Saruman*
- [ ] Something for running count-von-count
- [ ] 7 laptops for us + Saruman (of DICT, @Iepoev)
- [ ] 3 screens
- [ ] 2 barcode scanners for Saruman (of FK, @Procrat)

*Counting infrastructure*
- [ ] 26 batons
- [ ] AA-batteries
- [ ] 6-7 Gyrids (of geology, @Iepoev)
- [ ] Batteries for Gyrids (of geology, @Iepoev)
- [ ] Two tablets for manual counting (one of @Silox, one of @feliciaan)

*SMS system for DJ contest*
- [ ] Cell phone with working SIM card (@Procrat)
- [ ] Deadpool

*Other stuff*
- [ ] UPS (+ cables)
- [ ] Power cables
- [ ] Label machine
- [ ] Soldeerbout
- [ ] Battery tester

Procrat, 2016-04-18:
Staat nog niet klaar:
- [ ] 4G-router
- [ ] Wifi AP
- [ ] Verdeelstekkers

Iepoev, 2016-04-18:
current checkmarks signify wether or not these are prepared on the table

Procrat, 2016-04-20:
Delivered.

closed 2016-04-20 by Procrat