12urenloop/cvc #142: Site: add map with bus/tram from all faculties to yachtdreef

Procrat, 2016-04-06:


Procrat, 2016-04-09:
- **Overpoort**: bus 6 van Plateaustraat naar Europalaan
- **Sint-Pietersstation**: bus 9 van St.-Pietersstation naar Yachtdreef
- **Campus De Sterre**: bus 77 van Sterre naar St.-Pietersstation
- **Campus Coupure**: te voet
- **Site Dunant**: te voet
- **Ledeganck**: bus 9 van Krijgslaan naar Yachtdreef
- **Technicum**: bus 6 van Bagattenstraat naar Europalaan
- **Site Farmacie**: bus 9 van Zomerstraat naar Yachtdreef
- **Site Mercator**: te voet
- **Site Merelbeke- Diergeneeskunde**: bus 41 van Merelbeke Plataan naar Gent Yachtdreef
- **Site Proeftuinstraat**: bus 65 van Proeftuinstraat naar Koning Albertbrug
- **Site UZ**: bus 4 van UZ naar St.-Pietersstation + bus 9 nar Yachtdreef
- **Site Volderstraat**: bus 22 van Zonnestraat naar Bernard Spaelaan
- **Site Sint-pietersnieuwstraat**: bus 6 van Bagattenstrat naar Yachtdreef
- **Site Rommelaere**: bus 6 van Bijloke naar Europalaan

Silox, 2016-04-10:
I started something at https://www.google.com/maps/d/edit?mid=zRa63TBQtixw.k9xR5F4-59f4, but this takes hours. In any case too much time for what it's worth, our visitors are adults who probably know Ghent and/or are able to navigator their way through it.

We can put something on the website like:
Site Sterre | :footprints: | :bike: | :bus:
Site Mercator | :footprints: | :bike: | :bus:

where every icon links to the google maps location with a route. This will also deliver a nicer result than a Google maps layout with random colours on it.

Thoughts @Procrat?

Procrat, 2016-04-11:
Done.

closed 2016-04-11 by Procrat