12urenloop/cvc #166: make a 12urenloop Org?

Iepoev, 2016-09-12:
This repo is basically a bunch of seperate apps aggregated in a root folder. Wouldn't it be better to have this in a github org in seperate repos?

wschella, 2016-09-12:
Since communication will often be related, seperate repo's will disperse information.
See issues, PR comments, ...

Iepoev, 2016-09-12:
you can refer to other repos in comments, for example https://github.com/ZeusWPI/blokmap/issues/9

Procrat, 2016-09-12:
What would the benefits be?

Iepoev, 2016-09-12:
Every separate project having its own history, its own issues and its own PRs on its own branch network. we've got 16 issues and some don't have a clear indication to what project the belong. Branches are currently required to have a prefix that tells which project it belongs to. These aren't life-threatening issues, but it seems a logical change.

NoctuaNivalis, 2016-09-13:
But we all love merging branches :-(

iliclaey, 2017-11-27:
Is this something we might still consider, or should we just close this? It doesn't seem to be a bad idea to me.

werthen, 2017-11-28:
Bit of an outsider on this one, but I think it's a good idea.

becousae, 2017-11-28:
I agree

iliclaey, 2017-11-28:
Great! [Transferring](https://help.github.com/articles/transferring-a-repository-owned-by-your-personal-account/) the repo to an organisation shouldn't be too hard, as well as [splitting](https://help.github.com/articles/splitting-a-subfolder-out-into-a-new-repository/) each directory into it's own repo.

iasoon, 2017-11-28:
I'm in favour of moving the websites out of this repository, but I'd keep the entire CVC system in the same repository.

Personally and for no sound reason, I don't like the idea of moving this code out of the Zeus organisation, especially CVC.

becousae, 2017-11-28:
I think this could work. My main concern is the websites.

CVC has a more complex structure, with shared config files if I'm not mistaken. So that can stay together for me, in a Zeus Repo.

NoctuaNivalis, 2017-11-28:
Moving the websites sounds like a good idea, then I can listen to updates on just the interesting parts.

closed 2018-03-09 by iliclaey