12urenloop/cvc #177: Recover gyrid img

Iepoev, 2017-04-18:
@Procrat any idea how we can flash the gyrid image of last year to the memory cards?
it's 100MB so it must be compressed somehow. @NoctuaNivalis assumes that null-padding was removed.

Procrat, 2017-04-18:
We just used `dd` IIRC. 100MB shouldn't take that long, right?

Best of luck with the preparations!

Procrat, 2017-04-18:
You could also mount the image and copy the files, I guess. I don't know if that would be faster.

NoctuaNivalis, 2017-08-07:
I'm assuming this has been solved. Perhaps someone should post the solution here and close this before we forget it again.

closed 2018-03-09 by iliclaey

NoctuaNivalis, 2018-03-09:
Since this is closed, I'm assuming someone documented this?

becousae, 2018-03-09:
Because we are now working with the espresso bins, we are documenting all things we are doing.
We (hopefully) won't be using the gyrids again, so we won't be needing documentation for them.