12urenloop/cvc #179: Daemonize CountVonCount

Theci, 2017-04-27:
Could be handy to prevent a unintended ctrl-c from crashing the entire event.

NoctuaNivalis, 2017-04-27:
I wouldn't integrate this in count-von-count, when the same can be accomplished with
```
abduco -n count-von-count bash -c "count-von-count > logs"
tail -f logs 
```
or tmux/screen for the less fortunate.

I used to run count-von-count like this, because people like @nudded and @Silox were in the same room.

Silox, 2017-04-27:
> What is this nasty Haskell debugging output? Better stop this quick. `Ctrl-C`. There.

fbegyn, 2018-08-08:
Until this is documented somewhere or some kind of daemonisation is introduced, this issue remains open for me.
If the solution is to use sessions, it should be mentioned somewhere other then this issue.