12urenloop/cvc #184: Evaluation 2018

rien, 2018-04-25:
- A more reliable baton?
  - RFC, Laz0rz, https://www.pozyx.io/, 
- Camera's for manual count
  * Maybe even automatic recognition in the future?
- CVC better logging
  * Useful data extractions: measurements per gyrid, lap time count, ...
  * Log different input and output streams
  * No log file overwriting
  * Status page with heartbeats
- Gyrid reliability
  * Caching? (Causes teleportation)
  * Better way to know which gyrids are up or down
  * Reliable (documented) way to start or stop the gyrid service
  * NTP
  * Use switches instead of making a circle (power off one gyrid without disconnecting the rest)
- Manual count: rewrite in something simple but useful
  * Status update
  * Total view
  * Optional integration with boxxy
  * Importing to count-von-count
  * Allow manually updating manual count
  * Manual count logging: lap times, timestamps, ...
- Organisation
  * Have a manual count team ready
  * Keep complaining people outside
  * More computer screens to show logs on
  * Run CVC on an external computer
  * Prevent cables from being run over by trucks, cars or carrriages

iasoon, 2018-04-25:
Pozyx looks really cool, unfortunately it is also expensive =/

JulienMar, 2018-06-30:
There are a lot of companies that offer very accurate sport timings with RFID tags and mats (eg http://timingsense.com/en/), maybe look into renting one of those system if the cost is not too high?

NoctuaNivalis, 2018-07-02:
I'd think It'd be less work (I'm not even considering the cost) to just replace the bluetooth gyrids and batons with RFID tags and readers. An improvised espressobin reader could be plugged into the existing CVC code by outputting the old CSV protocol.

nudded, 2018-10-18:
You could also contact den dodentocht. They have 10 of these and they work with multiple people walking over them. You really only need 1. (perhaps 2 as a backup). Perhaps they will tell you how they did it and how much it costs. Or even better: send one of theirs.